tum-esm/pyra #149: Add a way to detect the router connectivity and an internal restart mechanism

patrickjaigner, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
Routers every now and then need to be manually restarted. This takes people to drive to the stations. 

**Describe the solution you'd like**
Research API commands and the HTTP interface of the router for direct communication for restart. 

**Describe alternatives you've considered**
Else it is also possible to do a power reset via a relay.

dostuffthatmatters, 2023-01-11:
Does a hard reset via the relay resolve the issue?

Then I would recommend doing that. Ping some public server every x hours and toggle the relay if the server doesn't answer.

patrickjaigner, 2023-01-11:
I think not all stations have a relay connected to the router. Only a subset has.
It would probably be elegant to talk to the router directly.

patrickjaigner, 2023-05-03:
There are two solutions to this problem. 

- (easy) Either the router offers an API request to trigger a restart
- (hard) We implement a power relay toggle in the PLC code directly

csam1853, 2023-05-04:
I rebooted the router via a Putty ssh connection with "root" as user and the router IP address (and our password).

closed 2024-05-14 by dostuffthatmatters