tum-esm/pyra #151: Unreachable Peer PLC issue since 4.0.7 on Mc04 and Md05

patrickjaigner, 2023-02-15:
First occurrence: 14.02.2023, 18:42

[image]

Traceback (most recent call last):

  File "C:\Users\EnclosureMc04\Documents\pyra\pyra-4.0.7\packages\core\main.py", line 116, in run
    m.run(config)

  File "C:\Users\EnclosureMc04\Documents\pyra\pyra-4.0.7\packages\core\modules\enclosure_control.py", line 91, in run
    self.plc_interface.connect()

  File "C:\Users\EnclosureMc04\Documents\pyra\pyra-4.0.7\packages\core\interfaces\plc_interface.py", line 140, in connect
    self.plc.connect(self.plc_ip, 0, 1)

  File "C:\Users\EnclosureMc04\AppData\Local\Programs\Python\Python310\lib\site-packages\snap7\client\__init__.py", line 24, in f
    check_error(code, context="client")

  File "C:\Users\EnclosureMc04\AppData\Local\Programs\Python\Python310\lib\site-packages\snap7\common.py", line 89, in check_error
    raise RuntimeError(error)

RuntimeError: b' TCP : Unreachable peer'

dostuffthatmatters, 2023-02-15:
This is probably due to `4.0.6` using `python-snap7@1.1` and `4.0.7` using `python-snap7@1.3`. This happened because we do automatic library upgrades with backwards compatible releases (here: `^1.1`). 

**Fix options:**
1. Fix the version at `1.1`.
2. Stay at `^1.1` and add resilience to the enclosure control procedure -> only raise an exception when the connection could not happen for three times in a row.

I prefer option 2, and we already implemented this resilience into the PLC communication:

https://github.com/tum-esm/pyra/blob/53e987fd1fbfa449cf71a5a6adc5e9c35c103e5e/packages/core/modules/enclosure_control.py#L158-L172

So, why do we now get `RuntimeExceptions` instead of `Snap7Exceptions`? Because they changed it. Here is the `1.1` implementation of the error-raising functions:

https://github.com/gijzelaerr/python-snap7/blob/3279ea1ca96b101021cb12890c929296e4724253/snap7/common.py#L67-L75

... and here is the `1.3` implementation:

https://github.com/gijzelaerr/python-snap7/blob/d26cfbe7da002c5527f5a1de549cc4c728640908/snap7/common.py#L75-L89

My change proposal:

```diff
 -  except Snap7Exception as e:
 +  except (Snap7Exception, RuntimeError) as e:
```

dostuffthatmatters, 2023-02-15:
In my opinion, this change in the `python-snap7` library a breaking one because it is not an accident. Hence, it should be a `2.0` (https://semver.org/).

patrickjaigner, 2023-02-16:
Good analysis. We should discuss if backwards compatible library updates of dependencies should be done automatically for our systems. 

I would strongly advice against it.
There are always risks for the instrument we can’t fully access and we don‘t have a indoor hardware setup to perform long duration tests for low occurrence errors.

commit a2cebac7 (2023-02-16): Close #151
commit 04e2b632 (2023-05-03): Close #151
commit 0559ecf1 (2023-05-03): Close #151
commit 70e8af75 (2023-05-09): Close #151
closed 2023-06-14 by dostuffthatmatters