tum-esm/pyra #155: DDE request failed on Ma and Mb

patrickjaigner, 2023-02-21:
**Describe the bug**
DDE requests continue to fail in every 2nd loop.

2023-02-21 10:40:32.128229 UTC+0 - opus-measurement - INFO - Running OpusMeasurement
2023-02-21 10:40:32.142196 UTC+0 - opus-measurement - INFO - Start OPUS.
2023-02-21 10:41:32.213668 UTC+0 - opus-measurement - INFO - Loading OPUS Experiment.
2023-02-21 10:41:32.550176 UTC+0 - opus-measurement - INFO - Connected to OPUS DDE Server.
2023-02-21 10:42:32.565005 UTC+0 - main - EXCEPTION - error occured: Traceback (most recent call last):

  File "C:\Users\ga56fem\Downloads\pyra\pyra-4.0.7\packages\core\main.py", line 116, in run
    m.run(config)

  File "C:\Users\ga56fem\Downloads\pyra\pyra-4.0.7\packages\core\modules\opus_measurement.py", line 78, in run
    self.automated_process_handling()

  File "C:\Users\ga56fem\Downloads\pyra\pyra-4.0.7\packages\core\modules\opus_measurement.py", line 283, in automated_process_handling
    self.load_experiment()

  File "C:\Users\ga56fem\Downloads\pyra\pyra-4.0.7\packages\core\modules\opus_measurement.py", line 147, in load_experiment
    answer = self.conversation.Request("LOAD_EXPERIMENT " + experiment_path)

dde.error: Request failed

2023-02-21 10:42:33.450810 UTC+0 - main - INFO - Ending iteration
2023-02-21 10:42:33.451808 UTC+0 - main - INFO - Starting iteration
2023-02-21 10:42:33.456795 UTC+0 - measurement-conditions - INFO - Running MeasurementConditions
2023-02-21 10:42:33.459787 UTC+0 - measurement-conditions - INFO - Helios as a trigger is considered.
2023-02-21 10:42:33.463776 UTC+0 - measurement-conditions - INFO - Measurements should be running is set to: True.
2023-02-21 10:42:33.466768 UTC+0 - enclosure-control - INFO - Running EnclosureControl
2023-02-21 10:42:41.274970 UTC+0 - enclosure-control - INFO - New continuous readings.
2023-02-21 10:42:41.329591 UTC+0 - sun-tracking - INFO - Running SunTracking
2023-02-21 10:42:42.504678 UTC+0 - opus-measurement - INFO - Running OpusMeasurement
2023-02-21 10:42:42.516682 UTC+0 - opus-measurement - INFO - Start OPUS.
2023-02-21 10:43:42.586549 UTC+0 - opus-measurement - INFO - Loading OPUS Experiment.
2023-02-21 10:43:42.977983 UTC+0 - opus-measurement - INFO - Could not connect to OPUS DDE Server.
2023-02-21 10:43:42.978980 UTC+0 - opus-measurement - INFO - DDE Connection seems to be not working.
2023-02-21 10:43:42.979978 UTC+0 - opus-measurement - INFO - Trying to fix DDE Socket.
2023-02-21 10:43:43.010456 UTC+0 - opus-measurement - INFO - Could not connect to OPUS DDE Server.
2023-02-21 10:43:43.051005 UTC+0 - system-checks - INFO - Running SystemChecks

**root cause**
There seems to be an issue with multiple instances of OPUS windows
displayed active OPUS instance: OPUS - Operator: {opus_username}  (Administrator) - [Display - default.ows1]

line of code in question:
opus_windows_name = (
            f"OPUS - Operator: {opus_username}  (Administrator) - [Display - default.ows]"

patrickjaigner, 2023-02-21:
Quick fix: Restart OPUS by restarting Pyra

patrickjaigner, 2023-05-03:
I will take care of this issue. I already have a solution for it in mind.

dostuffthatmatters, 2023-11-18:
This should be fixed by #202

commit d2762158 (2023-05-03): #155 Implement startup logic

Close #155