tum-esm/pyra #158: JOSS review

nmstreethran, 2023-03-27:
Hi, I am one of the reviewers assigned to your [submission to JOSS](https://github.com/openjournals/joss-reviews/issues/5131). My apologies for the delay in reviewing Pyra. Please find some comments regarding your submission below.

### Functionality / Documentation

- In the Setup guide (https://pyra.esm.ei.tum.de/docs/user-guide/setup):
  - There is a typo in the title: "**Hot** to set it up".
  - Can you add some information about how one can acquire and install OPUS/CamTracker? Are these software packages freely accessible? Provide links to their websites, if available.
- The [pyra-setup-tool](https://github.com/tum-esm/pyra-setup-tool) repo does not contain an open-source licence file.
- I suggest copying and pasting "What is Pyra?" from your documentation (https://pyra.esm.ei.tum.de/docs/) to this repo's README. The README currently only have links but having a brief overview of the software here would be good.

### Paper

- Statement of need
  - define GHG acronym in the second sentence
  - the acronym FTIR is used later in the text but isn't defined; this should probably be mentioned here?: "Ground-based remote sensing instruments that analyze direct sunlight using, for example, EM27/SUN Fourier transform spectrometers..."
- Summary
  - define the acronyms CLI, UI
- Bibliography: there are some formatting issues with the reference list in the PDF:
  - Most titles / book titles are not capitalised properly, e.g. see Che et al. (2022) where "beijing" is not capitalised; you can enclose the titles in double curly braces in the BibTeX file to fix this
  - Luther et al. (2022) - missing DOI: `10.5194/acp-22-5859-2022`
  - Kille et al. (2019) - `https://doi.org/` appears twice
  - Che et al. (2022) - subscript rendered incorrectly in the title (CO2)
  - Alberti et al. (2021) - the final author appears as "EM27/SUN-partners team, the"
  - Some author names have incorrect accents, for example Butz et al. (2017) (L"ubcke)
  - Journal titles are not consistent; most use full names but some use just the acronyms, for example Butz et al. (2017)

Many thanks!

patrickjaigner, 2023-04-04:
Dear @nmstreethran,
thank you for the detailed feedback. 

We updated all your points and fully rewrote the bibliography by downloading the Bibtex files from the journals directly.

For the definition of UI and CLI in the summary section, we were unsure how to improve it. Currently, they are introduced like this: Pyra Command Line Interface (Pyra CLI), and Pyra User Interface (Pyra UI). Maybe you have an idea on how to improve this?

nmstreethran, 2023-04-06:
Hi @patrickjaigner, thank you for addressing my points.

> For the definition of UI and CLI in the summary section, we were unsure how to improve it. Currently, they are introduced like this: Pyra Command Line Interface (Pyra CLI), and Pyra User Interface (Pyra UI). Maybe you have an idea on how to improve this?

I'll review your paper again and get back to you on this. I have some comments on the installation below.

I tried to set-up Pyra as detailed in the user guide on my system running Windows 11 and seem to have run into some issues.

`python run.py` executes as expected, but `status` shows no local Pyra versions and the CLI version points to `undefined`. The `Documents\pyra` directory just has an empty folder `ui-installers`. Only `colorama`, `pip`, and `setuptools` are installed in the system interpreter.

```
PS C:\Users\geog\Documents\pyra-setup-tool> python run.py
------------------------------------------------------------------------------------------------------------------------
Welcome to the pyra-setup-tool! Instruction formatting:
  * checkpoints = green
  * expecting user input = yellow
------------------------------------------------------------------------------------------------------------------------
Please DO NOT use a virtual environment for PYRA! Use the system interpreter available via the command 'python'. Currently using the interpreter 'C:\Users\geog\AppData\Local\Programs\Python\Python310\python.exe'
Run 'which python' in another shell. Are system-interpreter and the current one identical? (Y | n) Y
Python version 3.10.10 is supported
Running command "poetry --version"" in directory C:\Users\geog\Documents\pyra-setup-tool"
Found poetry!
Running command "tar --version"" in directory C:\Users\geog\Documents\pyra-setup-tool"
Found tar!
Running command "git --version"" in directory C:\Users\geog\Documents\pyra-setup-tool"
Found git!
Running command "gh --version"" in directory C:\Users\geog\Documents\pyra-setup-tool"
Found github cli!
Running command "git pull"" in directory C:\Users\geog\Documents\pyra-setup-tool"
Setup tool is up to date.
Created directory 'C:\Users\geog\Documents\pyra'
Created directory 'C:\Users\geog\Documents\pyra\ui-installers'
Successfully initialized local environment inside "C:\Users\geog\Documents\pyra"
------------------------------------------------------------------------------------------------------------------------
Enter a command (help | status | install | rollback | remove | exit) status
Did not find any local pyra versions.
Running command "gh release list --repo tum-esm/pyra"" in directory C:\Users\geog\Documents\pyra-setup-tool"
Remote pyra versions: 4.0.7, 4.0.6, 4.0.5
Pyra CLI found in environment PATH variable: Running command "echo %PATH%"" in directory C:\Users\geog\Documents\pyra-setup-tool"
No
Pyra CLI is pointing to version: undefined
------------------------------------------------------------------------------------------------------------------------
Enter a command (help | status | install | rollback | remove | exit)
```

So, I tried to run `install` but GitHub CLI has a bug (https://github.com/cli/cli/issues/7283) that's causing the installation to fail.

```
Enter a command (help | status | install | rollback | remove | exit) install
Running command "gh release list --repo tum-esm/pyra"" in directory C:\Users\geog\Documents\pyra-setup-tool"
Which version should be installed? (4.0.7 | 4.0.6 | 4.0.5) 4.0.7
Running command "gh release download --repo tum-esm/pyra --archive=tar.gz v4.0.7"" in directory C:\Users\geog\Documents\pyra"
Exception occured!
Traceback (most recent call last):
  File "C:\Users\geog\Documents\pyra-setup-tool\run.py", line 5, in <module>
    main.run()
  File "C:\Users\geog\Documents\pyra-setup-tool\src\main.py", line 107, in run
    raise e
  File "C:\Users\geog\Documents\pyra-setup-tool\src\main.py", line 88, in run
    commands.install.run()
  File "C:\Users\geog\Documents\pyra-setup-tool\src\commands\install.py", line 24, in run
    tasks.manage_local_files.download_version(version_to_be_installed)
  File "C:\Users\geog\Documents\pyra-setup-tool\src\tasks\manage_local_files.py", line 16, in download_version
    shell_utils.run_shell_command(
  File "C:\Users\geog\Documents\pyra-setup-tool\src\utils\shell_utils.py", line 33, in run_shell_command
    assert p.returncode == 0, (
AssertionError: command 'gh release download --repo tum-esm/pyra --archive=tar.gz v4.0.7' failed with exit code 2: stderr = 'panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x38 pc=0xc59337]

goroutine 9 [running]:
[... 25 lines cut ...]
        /opt/hostedtoolcache/go/1.19.7/x64/src/net/http/client.go:715 +0x8fc
net/http.(*Client).Do(...)
        /opt/hostedtoolcache/go/1.19.7/x64/src/net/http/client.go:581
github.com/cli/cli/v2/pkg/cmd/release/shared.fetchReleasePath({0x26bc4d8, 0xc0003e0040}, 0xc000883fa0?, {0xc00049f6c0?, 0x3?}, {0xc0004ce000, 0x27})
        /home/runner/work/cli/cli/pkg/cmd/release/shared/fetch.go:206 +0xe9
github.com/cli/cli/v2/pkg/cmd/release/shared.FetchRelease.func1()
        /home/runner/work/cli/cli/pkg/cmd/release/shared/fetch.go:140 +0x1a7
created by github.com/cli/cli/v2/pkg/cmd/release/shared.FetchRelease
        /home/runner/work/cli/cli/pkg/cmd/release/shared/fetch.go:138 +0x15e
'
PS C:\Users\geog\Documents\pyra-setup-tool>
```

I downgraded GitHub CLI from `2.26.1` to `2.25.1` but the issue still persists. Which version did you use?

I'm not sure why Pyra does not install when `python run.py` is executed. Am I missing any steps, or could this be due to the CLI bug?

Software versions:

```
Python 3.10.10
Poetry (version 1.4.2)
git version 2.40.0.windows.1
gh version 2.26.1 (2023-04-04)
```

nmstreethran, 2023-04-11:
@patrickjaigner I'm afraid I'm still having issues. See below. There still isn't a local Pyra version after running the setup tool.

Any idea why this isn't working on my side? I'm guessing the curl command piped into tar isn't working as expected.

```
PS C:\Users\geog> cd .\Documents\
PS C:\Users\geog\Documents> git clone https://github.com/tum-esm/pyra-setup-tool
Cloning into 'pyra-setup-tool'...
remote: Enumerating objects: 631, done.
remote: Counting objects: 100% (31/31), done.
remote: Compressing objects: 100% (21/21), done.
remote: Total 631 (delta 12), reused 10 (delta 10), pack-reused 600
Receiving objects: 100% (631/631), 114.16 KiB | 578.00 KiB/s, done.
Resolving deltas: 100% (363/363), done.
PS C:\Users\geog\Documents> cd .\pyra-setup-tool\
PS C:\Users\geog\Documents\pyra-setup-tool> python run.py
------------------------------------------------------------------------------------------------------------------------
Welcome to the pyra-setup-tool! Instruction formatting:
  * checkpoints = green
  * expecting user input = yellow
------------------------------------------------------------------------------------------------------------------------
Please DO NOT use a virtual environment for PYRA! Use the system interpreter available via the command 'python'. Currently using the interpreter 'C:\Users\geog\AppData\Local\Programs\Python\Python310\python.exe'
Run 'which python' in another shell. Are system-interpreter and the current one identical? (Y | n) Y
Python version 3.10.10 is supported
Running command "poetry --version" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Found poetry!
Running command "curl --version" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Found curl!
Running command "tar --version" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Found tar!
Running command "git --version" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Found git!
Running command "git pull" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Setup tool is up to date.
Created directory 'C:\Users\geog\Documents\pyra'
Created directory 'C:\Users\geog\Documents\pyra\ui-installers'
Successfully initialized local environment inside "C:\Users\geog\Documents\pyra"
------------------------------------------------------------------------------------------------------------------------
Enter a command (help | status | install | rollback | remove | exit) status
Did not find any local pyra versions.
Running command "curl --request GET --url "https://api.github.com/repos/tum-esm/pyra/releases" --header "Accept: application/vnd.github+json" --header "X-GitHub-Api-Version: 2022-11-28"" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Remote pyra versions: 4.0.7, 4.0.6
Running command "echo %PATH%" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Pyra CLI found in environment PATH variable: No
Pyra CLI is pointing to version: undefined
------------------------------------------------------------------------------------------------------------------------
Enter a command (help | status | install | rollback | remove | exit) install
Running command "curl --request GET --url "https://api.github.com/repos/tum-esm/pyra/releases" --header "Accept: application/vnd.github+json" --header "X-GitHub-Api-Version: 2022-11-28"" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Which version should be installed? (4.0.7 | 4.0.6) 4.0.7
Running command "curl -L https://github.com/tum-esm/pyra/archive/refs/tags/v4.0.7.tar.gz | tar zx" in directory "C:\Users\geog\Documents\pyra"
Exception occured!
Traceback (most recent call last):
  File "C:\Users\geog\Documents\pyra-setup-tool\run.py", line 5, in <module>
    main.run()
  File "C:\Users\geog\Documents\pyra-setup-tool\src\main.py", line 115, in run
    raise e
  File "C:\Users\geog\Documents\pyra-setup-tool\src\main.py", line 96, in run
    commands.install.run()
  File "C:\Users\geog\Documents\pyra-setup-tool\src\commands\install.py", line 41, in run
    tasks.download_version.download_version(version_to_be_installed)
  File "C:\Users\geog\Documents\pyra-setup-tool\src\tasks\download_version.py", line 12, in download_version
    utils.shell_utils.run_shell_command(
  File "C:\Users\geog\Documents\pyra-setup-tool\src\utils\shell_utils.py", line 41, in run_shell_command
    p.returncode == 0
AssertionError: command 'curl -L https://github.com/tum-esm/pyra/archive/refs/tags/v4.0.7.tar.gz | tar zx' failed with exit code 1: stderr = '  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0tar: Error opening archive: Failed to open '\\.\tape0'
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0 5215k    0 12842    0     0  17279      0  0:05:09 --:--:--  0:05:09 17279
curl: (23) Failure writing output to destination
'
PS C:\Users\geog\Documents\pyra-setup-tool>
```

nmstreethran, 2023-04-11:
@patrickjaigner I just tried it with Git Bash and it works now. I think it would be good if you state which shell to use in the docs (and also that the software is meant to be run on Windows). I mistakenly used PowerShell after seeing the tip about running `.ps1` scripts in the setup guide.

```
geog@DELL-LATITUDE-5410 MINGW64 ~
$ cd Documents/pyra-setup-tool/

geog@DELL-LATITUDE-5410 MINGW64 ~/Documents/pyra-setup-tool (main)
$ python run.py
------------------------------------------------------------------------------------------------------------------------
Welcome to the pyra-setup-tool! Instruction formatting:
  * checkpoints = green
  * expecting user input = yellow
------------------------------------------------------------------------------------------------------------------------
Please DO NOT use a virtual environment for PYRA! Use the system interpreter available via the command 'python'. Currently using the interpreter 'C:\Users\geog\AppData\Local\Programs\Python\Python310\python.exe'
Run 'which python' in another shell. Are system-interpreter and the current one identical? (Y | n) Y
Python version 3.10.10 is supported
Running command "poetry --version" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Found poetry!
Running command "curl --version" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Found curl!
Running command "tar --version" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Found tar!
Running command "git --version" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Found git!
Running command "git pull" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Setup tool is up to date.
Created directory 'C:\Users\geog\Documents\pyra'
Created directory 'C:\Users\geog\Documents\pyra\ui-installers'
Successfully initialized local environment inside "C:\Users\geog\Documents\pyra"
------------------------------------------------------------------------------------------------------------------------
Enter a command (help | status | install | rollback | remove | exit) status
Did not find any local pyra versions.
Running command "curl --request GET --url "https://api.github.com/repos/tum-esm/pyra/releases" --header "Accept: application/vnd.github+json" --header "X-GitHub-Api-Version: 2022-11-28"" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Remote pyra versions: 4.0.7, 4.0.6
Running command "echo %PATH%" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Pyra CLI found in environment PATH variable: No
Pyra CLI is pointing to version: undefined
------------------------------------------------------------------------------------------------------------------------
Enter a command (help | status | install | rollback | remove | exit) install
Running command "curl --request GET --url "https://api.github.com/repos/tum-esm/pyra/releases" --header "Accept: application/vnd.github+json" --header "X-GitHub-Api-Version: 2022-11-28"" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Which version should be installed? (4.0.7 | 4.0.6) 4.0.7
Running command "curl -L https://github.com/tum-esm/pyra/archive/refs/tags/v4.0.7.tar.gz | tar zx" in directory "C:\Users\geog\Documents\pyra"
Downloaded code
Running command "curl -L -O "https://github.com/tum-esm/pyra/releases/download/v4.0.7/Pyra.UI_4.0.7_x64_en-US.msi"" in directory "C:\Users\geog\Documents\pyra\ui-installers"
Downloaded UI
Running command "poetry config virtualenvs.create false" in directory "C:\Users\geog\Documents\pyra\pyra-4.0.7"
Running command "poetry env use system" in directory "C:\Users\geog\Documents\pyra\pyra-4.0.7"
Running command "poetry install" in directory "C:\Users\geog\Documents\pyra\pyra-4.0.7"
Skipping virtualenv creation, as specified in config file.
Installing dependencies from lock file

Package operations: 45 installs, 1 update, 0 removals

  • Installing pycparser (2.21)
  • Installing cffi (1.15.1)
  • Installing attrs (22.2.0)
  • Installing bcrypt (4.0.1)
  • Installing cryptography (39.0.1)
  • Installing exceptiongroup (1.1.0)
  • Installing iniconfig (2.0.0)
  • Installing numpy (1.24.2)
  • Installing packaging (23.0)
  • Installing pluggy (1.0.0)
  • Installing pynacl (1.5.0)
  • Installing six (1.16.0)
  • Installing tomli (2.0.1)
  • Installing click (8.1.3)
  • Installing coverage (7.1.0)
  • Installing mypy-extensions (1.0.0)
  • Installing invoke (1.7.3)
[... 26 lines cut ...]
  • Installing types-invoke (1.7.3.17)
  • Installing types-paramiko (2.12.0.3)
  • Installing types-psutil (5.9.5.6)
Connection pool is full, discarding connection: pypi.org. Connection pool size: 10
Installed code dependencies
Please install the UI using the installer that opens now
Running command "msiexec /i C:\Users\geog\Documents\pyra\ui-installers\Pyra.UI_4.0.7_x64_en-US.msi /qf" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Installed the UI
Updated the link in pyra-cli.bat
Running command "echo %PATH%" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Make the "pyra-cli" command available, by adding "C:\Users\geog\Documents\pyra" to your "user environment variables". See the pyra setup docs. (ok) ok
Created desktop shortcut to code directory
Skipping migration, no available versions to migrate from
done!
------------------------------------------------------------------------------------------------------------------------
Enter a command (help | status | install | rollback | remove | exit) status
Local pyra versions: 4.0.7
[... cut ...] --header "X-GitHub-Api-Version: 2022-11-28"" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Remote pyra versions: 4.0.7, 4.0.6
Running command "echo %PATH%" in directory "C:\Users\geog\Documents\pyra-setup-tool"
Pyra CLI found in environment PATH variable: No
Pyra CLI is pointing to version: 4.0.7
------------------------------------------------------------------------------------------------------------------------
Enter a command (help | status | install | rollback | remove | exit)
```

patrickjaigner, 2023-04-12:
@nmstreethran That's a very good point. We will add additional information for it to the documentation.

nmstreethran, 2023-04-12:
@patrickjaigner

I checked Pyra UI and Pyra CLI and everything is working well. Based on this and your suggestions in https://github.com/openjournals/joss-reviews/issues/5131#issuecomment-1493958647 re: the functionality, I've checked the remaining items in my checklist.

Some minor comments on your paper:

- There are still some issues with capitalisation in the bibliography (see the [PDF](https://raw.githubusercontent.com/openjournals/joss-papers/joss.05131/joss.05131/10.21105.joss.05131.pdf)). I've attached a file ([updated.txt](https://github.com/tum-esm/pyra/files/11210936/updated.txt)) with the suggested changes which you can review and incorporate into your BibTeX file.
- Under Software Dependencies, can you update the bit about the Pyra Setup Tool (since you're no longer using GitHub CLI)?
- Under Software Dependencies, you state that the UI is a Windows 10 application. I'm not sure if the Windows version is relevant here... I'm guessing I should be able to use it on Win 8/10/11 without any issues.
- In the Summary, under Pyra CLI, you state "The core commands allow direct interaction with the Pyra Core process and manage that only one instance of the Pyra Core process is running at the same time". If I understand this correctly, I think it will be clearer if you state "... **ensure** that only one instance of the Pyra Core process is running at **a** time."

Additionally:

- The link to the contributor guide in the repo's README is broken.
- Another thing that you may want to consider is adding a link to the docs and/or GitHub repo to the Pyra UI and CLI so users can easily get support. Note that this is just a suggestion and is not needed for the JOSS review.
- In the setup guide, I suggest changing the colorama installation command to `python -m pip install colorama` (which is [recommended](https://docs.python.org/3/installing/index.html)).

closed 2023-04-12 by nmstreethran