tum-esm/pyra #167: Some Helios cameras don't accept new settings

dostuffthatmatters, 2023-05-23:
[image]

Problem: this is blocking measurements
Solution: Only raise assertion errors on exposure setting (the most important) but only warn on the others.

dostuffthatmatters, 2023-05-23:
The solution is this: I will only log a warning, when the camera did not accept a particular setting. However, instead of only sending all settings at the initialization, I will now send all settings on every update of the exposure (every 3 minutes). With this, Helios will not block everything else just because the camera communication is not fully working, but will try it again every 3 minutes.

patrickjaigner, 2023-05-23:
This makes sense. A solution could also be a retry functionality, but I agree that a 3-minute cycle should be good enough and makes the code easier.

commit 9bc0d05e (2023-05-23): Close #167
PR #166 (2023-06-14): Integration 4.0.8
**Integration still in progress:**

- [x] Freeze the poetry dependencies to the one used in `4.0.7`
- [x] Remove git commit SHA from emails
- [x] Closes #151
- [x] Closes #126 
- [x] Closes #125 
- [x] Closes #154
- [x] Closes #157 
- [x] Closes #165
- [x] Closes #156 
- [x] Closes #167
- [x] Close #169
closed 2023-06-14 by dostuffthatmatters