tum-esm/pyra #169: Helios autoexposure is not properly working

patrickjaigner, 2023-05-27:
Station: Ma01
SW Version: 4.0.8
Time: 27.05.2023, 07.30 am UTC  

Helios indicates bad sun conditions (see picture)
Images are overexposed. 

Fix:
Restarting Pyra-core fixed the issue.

[image]

patrickjaigner, 2023-05-27:
After restart of Pyra core:
[image]

dostuffthatmatters, 2023-05-27:
The weird thing is, that the algorithm is doing what it is supposed to do. This is the result from running the autoexposure function at 07:36:19:

```json
[
    { "exposure": -13, "mean": 189.79 },
    { "exposure": -12, "mean": 189.79 },
    { "exposure": -11, "mean": 173.033 },
    { "exposure": -10, "mean": 125.409 },
    { "exposure": -9, "mean": 156.964 },
    { "exposure": -8, "mean": 210.211 },
    { "exposure": -7, "mean": 230.325 },
    { "exposure": -6, "mean": 249.346 },
    { "exposure": -5, "mean": 254.0 },
    { "exposure": -4, "mean": 254.0 },
    { "exposure": -3, "mean": 254.0 },
    { "exposure": -2, "mean": 254.0 },
    { "exposure": -1, "mean": 254.0 }
]
```

The algorithm picked exposure `-10`, which is the one it is supposed to pick.

The camera seems to be unstable when changing the exposure. I will implement some improvements.

dostuffthatmatters, 2023-05-27:
I made multiple fixes:

1. Wait 0.2s after setting exposure in autoexposure function
2. Fetch and throw away 5 images after changing exposure in exposure function (that has always been used when changing settings from anywhere else)
3. Fetch three images per exposure and average their mean colors
4. Wait 0.1s between fetching each image

In addition to that, the function will raise an Exception when the mean image colors are not increasing with exposure (this is the case here).

These changes should resolve all issues.

dostuffthatmatters, 2023-05-27:
This is how it is supposed to be (strictly ascending):

```json
[
    { "exposure": -13, "mean": 60.334 },
    { "exposure": -12, "mean": 60.088 },
    { "exposure": -11, "mean": 64.024 },
    { "exposure": -10, "mean": 107.313 },
    { "exposure": -9, "mean": 146.955 },
    { "exposure": -8, "mean": 189.574 },
    { "exposure": -7, "mean": 212.361 },
    { "exposure": -6, "mean": 241.18 },
    { "exposure": -5, "mean": 254.0 },
    { "exposure": -4, "mean": 254.0 },
    { "exposure": -3, "mean": 254.0 },
    { "exposure": -2, "mean": 254.0 },
    { "exposure": -1, "mean": 254.0 }
]
```

The autoexposure checks if each mean color is less than next-higher-exposure-color + 5.

dostuffthatmatters, 2023-05-27:
The new list of exposure results statically typed:

```py
[
    ExposureResult(exposure=-13, means=[105.65, 105.703, 105.566]),
    ExposureResult(exposure=-12, means=[130.778, 130.911, 130.961]),
    ExposureResult(exposure=-11, means=[164.475, 164.452, 164.443]),
    ExposureResult(exposure=-10, means=[211.698, 211.768, 211.759]),
    ExposureResult(exposure=-9, means=[248.78, 248.78, 248.795]),
    ExposureResult(exposure=-8, means=[252.786, 252.782, 252.788]),
    ExposureResult(exposure=-7, means=[253.999, 253.999, 253.999]),
    ExposureResult(exposure=-6, means=[254.0, 254.0, 254.0]),
    ExposureResult(exposure=-5, means=[254.0, 254.0, 254.0]),
    ExposureResult(exposure=-4, means=[254.0, 254.0, 254.0]),
    ExposureResult(exposure=-3, means=[254.0, 254.0, 254.0]),
    ExposureResult(exposure=-2, means=[254.0, 254.0, 254.0]),
    ExposureResult(exposure=-1, means=[254.0, 254.0, 254.0]),
]
```

PR #166 (2023-06-14): Integration 4.0.8
**Integration still in progress:**

- [x] Freeze the poetry dependencies to the one used in `4.0.7`
- [x] Remove git commit SHA from emails
- [x] Closes #151
- [x] Closes #126 
- [x] Closes #125 
- [x] Closes #154
- [x] Closes #157 
- [x] Closes #165
- [x] Closes #156 
- [x] Closes #167
- [x] Close #169
closed 2023-06-14 by dostuffthatmatters