tum-esm/pyra #170: CamTracker not closing down after manual stop (4.0.8)

patrickjaigner, 2023-05-27:
During my manual start and stops on Ma01 on 27.05.2023 I saw that CamTracker is not properly shutting down compared to version 4.0.7.

TODO: needs additional analysis.

dostuffthatmatters, 2023-05-27:
This is not an issue with either `4.0.7` or `4.0.8`, it has persisted since at least `4.0.6`: https://github.com/tum-esm/pyra/compare/v4.0.6...integration-4.0.8#diff-8c0d2fdeedda01ddf781e8e00a65657b6878dad56d1e9900dae1f961e20361ea

dostuffthatmatters, 2023-05-27:
I could not reproduce it. It was doing measurements in automatic mode, then I switched to manual mode, and then I clicked "stop". Then CamTracker closed.

@patrickjaigner Could you show me a video of that happening?

patrickjaigner, 2023-05-27:
I retested it. It worked for me now. Maybe I was too impatient. 

-> Closed without action

closed 2023-05-27 by patrickjaigner