tum-esm/pyra #181: [BUG]: Upload connection error causes Pyra to stall

csam1853, 2023-09-01:
### Descriptions

[image]

Due to an internet connection error of the VPN client, Pyra stopped to go any further except the upload-loop. Although conditions were bad, the cover opened up on its own because it followed the cam tracker.

It is also recommended, that the upload time should not overlap with the measurement times.

After ending and restarting the Pyra process and disabling the automatic upload it worked again.

### Pyra Version

4.0.7

dostuffthatmatters, 2023-09-01:
Interesting. For that reason the upload actually runs in a thread and should not affect the main loop (same as Helios).

I will investigate.

Thank you for the report.

closed 2023-10-25 by dostuffthatmatters

dostuffthatmatters, 2023-10-25:
Tracked in #193

dostuffthatmatters, 2023-10-25:
This likely happened, because the thread was somehow not set to run as a daemon. This has been fixed now. A non-daemon thread can continue its execution even after the main thread has exited.