tum-esm/pyra #187: Pyra cannot handle empty CamTracker log files

dostuffthatmatters, 2023-10-06:
### Descriptions

Issue #184 will be resolved by restarting CamTracker when it is not producing logs. This is not done by default but has to actively be enabled.

However when CamTracker never produces these logs, this _optional_ restart is not optional anymore because instead of resulting in "last log line is too old", it results in "cannot read last log line" which will _always_ trigger a restart of the CamTracker.

We can assume that all log files have _some_ lines in them otherwise, Pyra would restart CamTracker over and over again. Hence, I think we can interpret this "file is empty" to "last log line too old" to make this restart optional.

@patrickjaigner what do you think on that. This is not super urgent though.

### Pyra Version

4.0.*

patrickjaigner, 2023-10-09:
Can we ever have empty files? After the initial setup of an EM27 system and the parametrization of CamTracker on the system, there should be at least one run that saw the sun and, therefore have proper log files.

dostuffthatmatters, 2023-11-01:
Fair enough, we can not consider this case. It means that the user has not set the path to the CamTracker logs correctly.

closed 2023-11-01 by dostuffthatmatters