tum-esm/pyra #198: Instruments are measuring with cover closed

waterdropchen, 2023-11-05:
### Descriptions

Instruments are measuring with cover closed, at either sunny or cloudy conditions. Possible solution: check whether measurements are still onging when the cover is closed. If yes, stop the measurements. 

### Pyra Version

4.0.9

dostuffthatmatters, 2023-11-05:
Thank you for the report.

This should be fixed by #184.

dostuffthatmatters, 2023-11-07:
> Thank you for the report.
> 
> This should be fixed by #184.

Hi @waterdropchen,

I did not read the issue carefully. It does make a lot of sense that we do not start the Opus Macro until the cover is open. The commit fb1d811117a708679e9b963803412dae7194923d implements this logic now: If we have info on the cover angle, only run Opus, when the cover is open. Hence, this works _with_ and _without_ the TUM Enclosure.

Best,
Moritz

closed 2023-12-06 by dostuffthatmatters