tum-esm/pyra #240: [BUG]: 4.1.4 -> 4.2.0 could not migrate config

willmorrison1, 2025-02-22:
### Descriptions

I have a system with pyra 4.1.4 installed and now try to upgate to 4.2.0. At this stage of the install I have the following:

```
Installed code dependencies
Please install the UI using the installer that opens now
Running command "msiexec /i C:\Users\fitlet2\Documents\pyra\ui-installers\Pyra.UI_4.2.0_x64_en-US.msi /qf" in directory "C:\Users\Public\Documents\pyra-setup-tool"
Installed the UI
Updated the link in pyra-cli.bat
Running command "echo %PATH%" in directory "C:\Users\Public\Documents\pyra-setup-tool"
Make the "pyra-cli" command available, by adding "C:\Users\fitlet2\Documents\pyra" to your "user environment variables". See the pyra setup docs. (ok)
Created desktop shortcut to code directory
Should we reuse the config.json from a previously installed version? (no | 4.1.4) 4.1.4
Could not migrate config. The config of version "4.1.4" might be invalid: Could not perform config migration 4.1.4 -> 4.2.0: TypeError("'NoneType' object is not a mapping")
Exception occured!
Traceback (most recent call last):
  File "C:\Users\Public\Documents\pyra-setup-tool\run.py", line 5, in <module>
    main.run()
  File "C:\Users\Public\Documents\pyra-setup-tool\src\main.py", line 120, in run
    raise e
  File "C:\Users\Public\Documents\pyra-setup-tool\src\main.py", line 98, in run
    commands.install.run(prerelease=False)
  File "C:\Users\Public\Documents\pyra-setup-tool\src\commands\install.py", line 48, in run
    tasks.migrate_config.migrate_config(available_versions_to_migrate_from, version_to_be_installed)
  File "C:\Users\Public\Documents\pyra-setup-tool\src\tasks\migrate_config.py", line 27, in migrate_config
    _migrate_config_files(version_to_migrate_from, version)
  File "C:\Users\Public\Documents\pyra-setup-tool\src\tasks\migrate_config.py", line 53, in _migrate_config_files
    json.dump(new_config, f)
UnboundLocalError: local variable 'new_config' referenced before assignment
PS C:\Users\Public\Documents\pyra-setup-tool>
```

the above upgrade feature doesn't seem to have worked. The 4.2.0 pyra instance then loads and there is no config file and a range of errors

[image]

no config

[image]

### Pyra Version

4.1.4 upgrading to 4.2.0

### Your `config.json` file

```json

```

willmorrison1, 2025-02-22:
I removed then reinstalled 4.2.0 and confirm that the following option is working ok (pyra UI is working, shows config UI, no error messages, but obviosuly the config has defaults)

```
Created desktop shortcut to code directory
Should we reuse the config.json from a previously installed version? (no | 4.1.4) no
done!
```

If you need more details then I can reproduce with another system that is untouched on 4.1.4

dostuffthatmatters, 2025-02-22:
Hi @willmorrison1,

thank you for the report! Apologies for the bug. I could reproduce it and just pushed the fix to the setup tool. It was caused because we only tested the migrations with all optional fields being present too.

willmorrison1, 2025-02-24:
thanks very much. I will let you know here when this is successful when I run this on another PC

dostuffthatmatters, 2025-05-28:
Hi @willmorrison1,

have you set up a new system by now?

dostuffthatmatters, 2025-05-28:
Great thanks :)

closed 2025-05-28 by dostuffthatmatters