tum-esm/pyra #262: [BUG]: PyraUI can not find backend, though backend can be found through command line interface

RobbieRamsay, 2025-07-22:
### Descriptions

I have installed Pyra correctly on to a Windows device. I can see that the Pyra installation is located in the correct folder. I have added the CLI to the Path enviroment variable, and can run that through Powershell. 
However, the PyrUI, once started, states: "PYRA has not been foound on your system...PYRA4 should be located at [path name to a Pyra installation, which exists on the device]". 

### Pyra Version

4.2.0

### OPUS Version

8.2

### CamTracker Version

3.9.2

### Windows Version

10

### Your `config.json` file

```json

```

dostuffthatmatters, 2025-07-22:
Hi @RobbieRamsay

thank you for reporting this issue! Could you take a screenshot of the error in the console of the UI window. Here are the instructions to get to that console: https://github.com/tum-esm/pyra/wiki/Setup-Issues#how-to-get-to-the-ui-console

RobbieRamsay, 2025-07-22:
Hello @dostuffthatmatters, thanks for the swift reply. Screenshot attached, and string output of stderr below:

[image]

C:\Users\fitlet2\AppData\Local\Programs\Python\Python310\lib\site-packages\snap7\__init__.py:4: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.

  import pkg_resources

Usage: pyra-cli [OPTIONS] COMMAND [ARGS]...

Options:

  --help  Show this message and exit.

Commands:

  config

  core

  info              Print Pyra version and code directory path.

  logs

  plc

  remove-filelocks  Remove all filelocks.

  state

  test

RobbieRamsay, 2025-07-22:
That worked. Thank you very much for the swift and helpeful resolution to this.

dostuffthatmatters, 2025-07-22:
Great :) sure! Feedback is helpful, then we can add notes to the docs.